billmania/roboquest_ui #30: Stop the topicPeriodS event emits when the socket is not connected

billmania, 2023-08-26:
For widgets, such as joystick and slider, which specify the topicPeriodS configuration attribute to continually emit events from the browser to the NodeJS server on the socket.io connection, only emit the events when the socket is connected with the NodeJS server. Don't start emitting until the socket is connected. Stop when the connection is broken, Resume when the connection is restored.

To reproduce, start the complete backend, connect the browser, verify the events are emitted, then shutdown the backend.

SuddenDevelopment, 2023-08-26:
1.  topicPeriodS event emits when the socket is not connected
2.  console log the disconnect
3. change the screen back to "Waiting for Robot"

SuddenDevelopment, 2023-08-30:
the joystick and slider repeaters were removed and we decided to not put them back in

closed 2023-09-05 by billmania