billmania/roboquest_ui #31: Slider widget doesn't repeat the current slider value

billmania, 2023-08-26:
Moving the slider doesn't cause the current slider value to be published until the mouseUp event. The current slider value is emitted once and then the interval resumes emitting what looks like the default slider value, even though the slider button is not at that value.

To reproduce, start the backend, start the frontend, connect the browser, use **ros2 topic echo /servos** to display the angle values. Move the slider without a mouseUP and note the published value doesn't change. Cause the mouseUp event and note a single published message at the current value after which the default value is continually published.

billmania, 2023-08-28:
The payload emitted onto the socket is malformed AND it is not being sent periodically.The malformed payload is
```
["angle,name":"178"]
```

SuddenDevelopment, 2023-08-29:
duplicate, merged to 26

closed 2023-08-29 by SuddenDevelopment