billmania/roboquest_ui #48: How to enable an added widget

billmania, 2023-08-28:
After the add/widget feature is used to add a widget to the display or to change the definition of a widget, what happens next? How does the widget become fully enabled? How does its configuration become persistent?

1. Automatically save the configuration
2. Request the user to explicitly save
3. Manually restart the software via a UI element
4. Manually reboot the robot
5. Saving the configuration automatically causes the container to restart and the browser to reload

SuddenDevelopment, 2023-08-29:
you previously said you wanted an explcit save action.
there is no auto save. The next explciti save from menu should include it.

I see what you mean though, because it's not just a UI element it needs to be read and processed.

my idea:
when UI saves config...

1. json recieved by middleware.
2. compare the config partts that matter to tell if subscriptions have changed requiring action.
3. restart any services required.
4. if needed, send a signal through websocket the UI can use to know it's time to reset the UI to coordinate. I dont think this should be needed though, when subscriptions are update it should just start getting those events.

billmania, 2023-08-29:
I agree with the basics of the earlier proposal. As discussed, we'll implement this feature in phases.

1. For v13, after the user makes the configuration changes, the browser UI Save Config button will be clicked. Any topic or service configuration changes will require a manual restart of the robot.
2. For a release after v13, an option will be added to the Configuration Settings to "restart the robot".
3. For a release after phase 2, communications will be defined between the robot and the browser UI, whereby several things will be implemented:
    a. the browser UI can ask the NodeJS server for a list of available topics and services
    b. the browser UI can ask the NodeJS server for the interface of a particular topic or service
    c. the browser UI can inform the NodeJS server of a configuration change by sending two objects
        i. the new widget's definition
        ii. the deleted widget's definition

The proposal above will be handled by this phase 3.

closed 2023-11-03 by billmania