billmania/roboquest_ui #64: Add keyboard options for the joystick widget P1

billmania, 2023-08-31:
Add the option to add keyboard events to produce the same effect as joystick options. This will allow a student to drive the robot using the keyboard instead of the joystick widget.

The defined keyboard events must be saved in the configuration for the joystick. Using the browser UI it must be possible to assign a keyboard event for each of:

- forward min speed, forward half speed, forward max speed
- reverse min speed, reverse half speed, reverse max speed
- turn left at min, half, and max speed
- turn right at min, half, and max speed
- stop all motors
Except for the "stop all motors" key, generating a keyboard event will cause that action for N seconds and then stop.

SuddenDevelopment, 2023-08-31:
try to capture the key event to configure. Including key combos?

SuddenDevelopment, 2023-09-01:
within the joystick widget:

- create a button to switch the joystick widget to a keyboard input "mode"
- create a number of named functions for each distinct event such as "left", "right, "forward", "medium speed", "high speed"
- set each of events from the joystick config to a key
- use jquery keydown event: https://api.jquery.com/keydown/
- speed + direction key would need to be a keypress at the same time to work.
- the resulting values need to set the same joystick events and values, triggering the same emit mechanism the joystick uses. This will mean it keeps the same data scaling etc.

billmania, 2023-09-01:
1. agreed
2. agreed
3. agreed, including some experimentation to determine which keys can be mapped to events and can bi visually represented on the joystick widget. ie. It will be challenging to select Ctl-PgDn-F11 as the "go faster" key and display it on the widget
4. agreed, with some prototyping to ensure its limits are understood
5. does this mean the separate "speed" key would be a modifier for each separate "direction" key?
6. strong agreement
Thinking more about 5. assume there's exactly one key defined for each of the four directions and a fifth key defined for full-stop. Also, assume there is no "don't release the key and go faster". Then, press the key to start moving and press it again to incrementally increase the speed. Then, press the full-stop key to stop or repeatedly press the opposite direction key to incrementally slow. Keep pressing it to reverse direction.

SuddenDevelopment, 2023-09-01:
5. yes: direction key will always be low speed. then a speed key would need to be added to increase speed in that direction

billmania, 2023-09-05:
**Implementation plan**

1. research the jQuery keyboard event capture mechanism
2. add a button to the top of the page which enables and disables "keyboard driving mode"
3. in keyboard mode, capture key events and log them
4. identify/define the functions which are used by joystick motion commands and can be associated with key events
5. map the key events to the joystick motion command functions

billmania, 2023-09-08:
Change the collection of keys and the way they're handled. Eliminate the STOP key. On keydown, start the motion. On keyup, stop the motion. Implement speed control via a separate slider enabled at all times (ie. it affects the joystick AND the keys).

closed 2023-09-12 by billmania