billmania/roboquest_ui #158: Multiple gamepad buttons controlling a single servo

rom1011, 2025-01-21:
I am trying to use the D-Pad on a gamepad for servos and I can get one button to work by using attribute=servo0.angle_incr_deg;servo0.command_type:2 and scaling=1. But if I add the same attribute=servo0.angle_incr_deg;servo0.command_type:2 and scaling=-1 to a second button, then only one button works (the one that has a higher gamepad ID). It won't work (servos do not move) if I use a value for the attribute like in attribute=servo0.angle_incr_deg:1;servo0.command_type:2

In the wGamepad.js _handleTopic() function I see this header
   * If this topic is assigned to a button, ie. the first
   * attribute includes a constant, only build the payload
   * if action.value is non-zero.
   * Otherwise always build the payload for any action.value.

I think this is happening: since I do not have a constant, the zero value of the button that is not pressed is not ignored and thus is overwriting the button that is pressed.

Is there any way to get two gamepad buttons to control one servo in angle_incr_deg mode?

billmania, 2025-01-21:
@rom1011 Contact me directly and I'll be happy to help. Include your name and your school affiliation.

billmania, 2025-01-21:
To resolve this issue, there are several requirements, some of which are already implemented in rq_ui v35.1:

1. any topic/attribute combination can be assigned to 0 or more buttons and/or axes
2. any service/attribute combination can be assigned to 0 or more buttons and/or axes
3. the value of the axis is assigned to the first attribute, iff the first attribute doesn't have an associated constant
4. when the same topic/attribute is assigned to more than one button or axis, only the pressed button or the non-zero axis value will be used
5. any configured gamepad axis will continue to publish its "released" value periodically
6. any attribute can have an associated constant, separated from its attribute by a colon
7. when more than one attribute is assigned, they are separated by a semi-colon
8. when more than one attribute is assigned, all but the first attribute must include a constant
9. attribute constant types must match the type of their associated attribute
10. when a gamepad button or axis is configured for a topic or service and a widget with a non-zero topic period has the same topic/service attribute combination, the widget will be disabled when the gamepad is enabled

billmania, 2025-02-05:
This issue has been resolved by

- [Issue 160](https://github.com/billmania/roboquest_ui/issues/160)
- [Issue 161](https://github.com/billmania/roboquest_ui/issues/161)
- [Issue 162](https://github.com/billmania/roboquest_ui/issues/162)

closed 2025-02-05 by billmania